openhwfoundation/cvw #1082: ImperasDV mismatch rv32gc WALLY-trap test

jordancarlin, 2024-11-10:
Was attempting to run the rv32gc wally-riscv-arch-tests in lockstep like we currently do for the rv64gc tests, and got the following mismatch on WALLY-trap-01.S.

```
wsim rv32gc tests/riscof/work/wally-riscv-arch-test/rv32i_m/privilege/src/WALLY-trap-01.S/ref/ref.elf --sim questa --lockstep
```

```
# Info (IDV) Instruction executed prior to mismatch '0x80000708(trap_handler_end_m+34): 31ae9073 csrrw   x0,menvcfgh,x29'
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x80000708 trap_handler_end_m+34):
# Error (IDV) Mismatch 0> CSR 31a (menvcfgh)
# Error (IDV)   . dut:0x00000000 STCE:0 PBMTE:0 (not updated)
# Error (IDV)   . ref:0x80000000 STCE:1 PBMTE:0
# Error (IDV) testbench.idv_trace2api.state_compare @ 3380: MISMATCH
```

jordancarlin, 2024-11-10:
It is actually many of the rv32gc privileged tests that fail in lockstep. It seems plausible that this could be a configuration issue with ImperasDV since the rv32gc lockstep configuration is newer.

- trap
- trap-s
- trap-u
- mmu
- pmp
- wfi
- lrsc
- mtvec.

davidharrishmc, 2024-11-10:
Not surprising to have some mismatches against privileged tests.  We had quite a few when we first ran rv64gc in lockstep.

This one looks like it might be a configuration issue about Sstc support/enabling

jordancarlin, 2024-11-10:
Looks like Sstc is supported in the rv32gc config and is enabled in the rv32gc imperas.ic file

davidharrishmc, 2024-11-13:
pmpcfg failing with

```
wsim rv32gc tests/riscof/work/wally-riscv-arch-test/rv32i_m/privilege/src/WALLY-pmp-01.S/ref/ref.elf --sim questa --lockstepverbose

# Info 289: 'refRoot/cpu', 0x00000000800008d4(write_pmpcfg_1): Machine 3a1e9073 csrrw   x0,pmpcfg1,x29
# Info   MEMX 0x800008d4 0x800008d4 2 9073
# Info   MEMX 0x800008d6 0x800008d6 2 3a1e
# Info (RISCV_PMP) CPU 'refRoot/cpu': PMP PRIV=--- 0x80100400:0x801005ff (mode Supervisor)
# Info (RISCV_PMP) CPU 'refRoot/cpu': PMP PRIV=--- 0x80100300:0x80100303 (mode Supervisor)
# Info (RISCV_PMP) CPU 'refRoot/cpu': PMP PRIV=--- 0x80100300:0x80100303 (mode Machine)
# Info (RISCV_PMP) CPU 'refRoot/cpu': PMP PRIV=--- 0x80100200:0x8010020f (mode Supervisor)
# Info (RISCV_PMP) CPU 'refRoot/cpu': PMP PRIV=--- 0x80100200:0x8010020f (mode Machine)
# Info   pmpcfg1 00000000 -> 0018900c
# Info   pmp4cfg1 00 -> 0c [L:0 A:0(OFF)->1(TOR) priv:0(---)->4(--x)]
# Info   pmp5cfg1 00 -> 90 [L:0->1 A:0(OFF)->2(NA4) priv:0(---)]
# Info   pmp6cfg1 00 -> 18 [L:0 A:0(OFF)->3(NAPOT) priv:0(---)]
# Info (IDV) Instruction executed prior to mismatch '0x800008d4(write_pmpcfg_1+0): 3a1e9073 csrrw   x0,pmpcfg1,x29'
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x800008d4 write_pmpcfg_1+0):
# Error (IDV) Mismatch 0> CSR 3a1 (pmpcfg1)
# Error (IDV)   . dut:0x00000000 (not updated)
# Error (IDV)   . ref:0x0018900c
# Error (IDV) testbench.idv_trace2api.state_compare @ 7260: MISMATCH
```

davidharrishmc, 2024-11-13:
and stimecmph failing with

```
wsim rv32gc tests/riscof/work/wally-riscv-arch-test/rv32i_m/privilege/src/WALLY-wfi-01.S/ref/ref.elf --sim questa --lockstepverbose

# Info 157: 'refRoot/cpu', 0x00000000800003ec(time_interrupt_m+10): Machine 15d39073 csrrw   x0,stimecmph,x7
# Info   MEMX 0x800003ec 0x800003ec 2 9073
# Info   MEMX 0x800003ee 0x800003ee 2 15d3
# Info   stimecmph 00000000 -> ffffffff
# Info (IDV) Instruction executed prior to mismatch '0x800003ec(time_interrupt_m+10): 15d39073 csrrw   x0,stimecmph,x7'
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x800003ec time_interrupt_m+10):
# Error (IDV) Mismatch 0> CSR 15d (stimecmph)
# Error (IDV)   . dut:0x00000000 (not updated)
# Error (IDV)   . ref:0xffffffff
```

davidharrishmc, 2024-11-13:
Common pattern is the DUT is always holding 0 and the ref is nonzero.

jordancarlin, 2024-11-13:
Yes. Similar issues for all of tests I listed above. Haven't been able to investigate much yet since license was down. Will be looking into it more soon.

jordancarlin, 2024-11-13:
Wally doing nothing while ref does something makes me think it may be a feature that should be disabled in  the ref, but I haven't found anything wrong in the configuration file yet.

davidharrishmc, 2024-11-13:
I wonder if the tracer is not working properly.

jordancarlin, 2024-11-13:
Interesting. That could explain it. Would be weird that only a few of the CSRs are mismatching if that’s the problem though.

davidharrishmc, 2024-11-13:
In testsbench/comm/wallyTracer.sv, at lines 160-205, the csr _REGW signals are copied into CSRArray.  Some registers such as 31A (MENVCFGH) only show up here.  Others such as 0x105 show up 11 more times in the file.  I think we added some registers (particularly RV32 H registers) and didn't update all the places they need to be copied.  @jordancarlin does this give you enough info to track them down?

Also, I think line 203 is a duplicate of 161 and probably can be deleted.

jordancarlin, 2024-11-14:
Making progress here. Many of the 32-bit only CSRs were not fully implemented in the tracer. I've fixed most of these and almost all of the tests are passing now.

The PMP CSRs are implemented slightly differently from the others. I still need to get their 32-bit version working.

lrsc is failing with a mismatch in a standard integer register. Still need to debug that one.

```
$ wsim rv32gc tests/riscof/work/wally-riscv-arch-test/rv32i_m/privilege/src/WALLY-lrsc-01.S/ref/ref.elf --sim questa --lockstepverbose

# Info 16: 'refRoot/cpu', 0x000000008000003c(rvtest_entry_point+3c): Machine 18bfa62f sc.w    x12,x11,(x31)
# Info   MEMX 0x8000003c 0x8000003c 2 a62f
# Info   MEMX 0x8000003e 0x8000003e 2 18bf
# Info   MEMW 0x80002000 0x80002000 4 0000002b
# Info   LRSCAddress 80002000 -> ffffffff
# Info (IDV) Instruction executed prior to mismatch '0x8000003c(rvtest_entry_point+3c): 18bfa62f sc.w    x12,x11,(x31)'
# Error (IDV) GPR register value mismatch (HartId:0, PC:0x8000003c rvtest_entry_point+3c):
# Error (IDV) Mismatch 0> GPR x12
# Error (IDV)   . dut:0x00000001
# Error (IDV)   . ref:0x00000000
# Error (IDV) testbench.idv_trace2api.state_compare @ 1150: MISMATCH
```

jordancarlin, 2024-11-14:
Found a solution for PMP. Kind of messy because of the loops, but I think it'll work. Will open PR soon.

jordancarlin, 2024-11-14:
#1091 fixes the tracer issues. The lrsc mismatch seems to be something different. Will open a different issue for that.

closed 2024-11-14 by jordancarlin