openhwfoundation/cvw #1086: Wally on Genesys 2 FPGA board?

matsbror, 2024-11-12:
Would it be difficult to set upp support for the [Genesys 2](https://digilent.com/reference/programmable-logic/genesys-2/start) board?

rosethompson, 2024-11-12:
Hi Mats Brorsson. It shouldn't be too much effort to port Wally to the Genesys 2 board. I just got my hands on one of these boards last week and I am hoping to work on this later this week or next.  If you are interested helping  the main things which are required are as follows.

1. New device tree to define clock speed and memory size.
2. New pin mappings for all the I/O devices.
3. Creating a new script to generate the Xilinx IP memory controller. I think we will be able to share the same memory controller script between the Arty A7 and the Genesys 2 board by parameterizing the memory part, size, bus width, etc.

matsbror, 2024-11-12:
Thanks @rosethompson! I currently am not able to help out with this until, hopefully, in January/February.

PietDelaney, 2025-05-18:
Genesys 2 FPGA board. Looks like a nice board, but kinda expensive for a student.
The Xilinx ML605 board has been much cheaper on eBay, but I fear the tools aren't available from AMD.

-piet

closed 2025-05-30 by davidharrishmc