openhwfoundation/cvw #1108: SPI code cleanup

davidharrishmc, 2024-11-16:
In spi_apb.sv:

* Entry and Memwrite should be entry and memwrite to be consistent with the other SPI modules.
* Why is this MemWrite case split in two?  If there is a good reason, please comment.  Otherwise, please merge.

      if (Memwrite)
        case(Entry) // flop to sample inputs
          SPI_SCKDIV:  SckDiv <= Din[11:0];
          SPI_SCKMODE: SckMode <= Din[1:0];
          SPI_CSID:    ChipSelectID <= Din[1:0];
          SPI_CSDEF:   ChipSelectDef <= Din[3:0];
          SPI_CSMODE:  ChipSelectMode <= Din[1:0];
          SPI_DELAY0:  Delay0 <= {Din[23:16], Din[7:0]};
          SPI_DELAY1:  Delay1 <= {Din[23:16], Din[7:0]};
          SPI_FMT:     Format <= {Din[19:16], Din[2]};                    
          SPI_TXMARK:  TransmitWatermark <= Din[2:0];
          SPI_RXMARK:  ReceiveWatermark <= Din[2:0];
          SPI_IE:      InterruptEnable <= Din[1:0];
        endcase

      if (Memwrite)
        case(Entry)
          SPI_TXDATA:  if (~TransmitFIFOFull) TransmitData[7:0] <= Din[7:0];
        endcase

davidharrishmc, 2024-11-16:
@JacobPease can you take a look at this one?

JacobPease, 2024-11-16:
> @JacobPease can you take a look at this one?

I think that's something left from trying to fix the previous SPI code. At one point the conditions were different. I can fix that and add it to my current PR.

JacobPease, 2024-11-16:
@davidharrishmc Fixed. Still passes regression too.

davidharrishmc, 2024-11-16:
Speedy!