openhwfoundation/cvw #1114: Difficulties getting RV64I working in lockstep

davidharrishmc, 2024-11-17:
Trying to get RV64I working in lockstep.  
Copied config/rv64gc/imperas.ic into config/rv64i

```
wsim rv64i tests/riscof/work/riscv-arch-test/rv64i_m/I/src/addi-01.S/ref/ref.elf --lockstepverbose
```

Immediate PC mismatch
```
# Error (IDV) PC mismatch (HartId:0, PC:0x0000000080000000 rvtest_init+0):
# Error (IDV) Mismatch 0>
# Error (IDV)   . dut:0x0000000000000000 
# Error (IDV)   . ref:0x0000000080000000 rvtest_init+0
```

@rosethompson looks like an initialization issue.  Do you have any suggestions?

jordancarlin, 2024-11-26:
Seems to be that as soon as Zicsr is disabled Imperas is no longer able to get the PC.

davidharrishmc, 2024-11-26:
The signals in wallyTracer.sv might give some clues, but I don't immediately see a signal gated by Zicsr.

davidharrishmc, 2025-04-03:
Using rv32imc instead.

closed 2025-04-03 by davidharrishmc