openhwfoundation/cvw #1120: Code coverage reporting higher coverage than expected

jordancarlin, 2024-11-21:
Code coverage reports that many sections of code that are not tested are covered. For example, even if crypto tests are not run, the code coverage report shows >99% of the crypto unit being covered.

This is likely because many of the signals are toggling even when a different instruction is run. Most units are not gated based on the currently executing instruction, so this is not all that surprising. There are a few weird cases though. For example, the bit manipulation unit has the following that should keep its inputs to zero when not targeting a bit manipulation instruction, yet still has fairly high coverage without running any bit manip tests.

```verilog
  // gate data inputs to BMU to only operate when BMU is active
  assign ABMU = A & {P.XLEN{BMUActive}};
  assign BBMU = B & {P.XLEN{BMUActive}};
```

This might end up all being an inevitable result of how Questa collects coverage, but it is worth looking into.

jordancarlin, 2024-12-02:
Code coverage was updated to use the cvw-arch-verif suite instead of riscv-arch-test. The same issue with unexpectedly high coverage is still occurring.

davidharrishmc, 2025-04-02:
@jordancarlin are you satisfied to close this after your experiments?

jordancarlin, 2025-04-02:
Seems to be a result of not gating most of our execution units, so the high coverage with random values is unavoidable. Closing this.

closed 2025-04-02 by jordancarlin