openhwfoundation/cvw #1121: Review need for wally-riscv-arch-test priv tests with cvw-arch-verif

jordancarlin, 2024-11-21:
Now that privileged tests are coming up in cvw-arch-verif, many of the wally-riscv-arch-test priv tests are/will be redundant and can probably be removed. Ones that come to mind already are CSRs and endianness, and that list will likely grow as more tests are developed. 

It is probably worth doing a review of all of these tests to see which need to be retained (mostly peripherals and a few other Wally specific ones I assume) and which can be trimmed down/removed entirely.

rosethompson, 2024-11-21:
Until we have an open source way to validate the execution of the cvw-arch-verif tests I'd like to keep our regression tests.  Not everyone has access to ImperasDV and I think there is value in keeping the old tests at least for now.

I see two ways around this either an open source lock step or self checking tests.

jordancarlin, 2024-11-21:
The fact that they are not yet possible to run with open source tools only is a very good point that I totally missed. Something to think about eventually then, but seemingly not yet.

jordancarlin, 2024-11-26:
Closing for now because we don't have a way to run those tests with open source tools yet. Will revisit down the road.

closed 2024-11-26 by jordancarlin