openhwfoundation/cvw #1128: Add Dependabot to keep submodules up to date

jordancarlin, 2024-11-25:
We rely on a variety of submodules, and for several of them almost always want to be on the latest version (cvw-arch-verif, riscv-arch-test, etc). Dependabot is a feature built into GitHub that can automatically open PRs when there is a new version of a submodule (or other dependency). It's used by many other projects, including CVA6. 

For simple submodules like cvw-arch-verif, we can probably just merge the PRs it creates as they are. For more complicated ones like riscv-arch-test it will at least alert us that there is a newer version and we can add other changes that are necessary manually.

See https://docs.github.com/en/code-security/dependabot/dependabot-version-updates

davidharrishmc, 2024-11-25:
Sounds promising.  Try it out.  If you don't like it, we can always roll back.

jordancarlin, 2024-11-25:
Done with PR #1129

jordancarlin, 2024-11-25:
Seems like it opened a bunch of PRs. We should review which ones we want to merge and which submodules should stay pinned to the current version. We can explicitly exclude any submodules we don't want updated in the Dependabot file.

jordancarlin, 2024-11-25:
Also a good chance to review if we actually still need/want all of these submodules.

jordancarlin, 2024-11-26:
Done with PR #1129. Will need modifications to avoid updating submodules that we want pinned to specific versions.

closed 2024-11-26 by jordancarlin