openhwfoundation/cvw #1146: Imperas uses wrong encoding for zip and unzip

davidharrishmc, 2024-11-28:
ImperasDV does not recognize the encodings of zip and unzip.  For example, per the latest build of the ISA manual, 
unzip x0, x0 = 08f05013
zip x28, x27 = 08fd9e13
In particular, instr[24:20] should be 01111.
GCC and Sail use this encoding.

The unpriv-isa-asciidoc_20240411.pdf contained incorrect encodings with 11110 for zip and 11111 for unzip.  It appears ImperasDV
may be using these incorrect encodings, and thus does not recognize 08f05013 as unzip.

```
 Info 23: 'refRoot/cpu', 0x0000000080005014(main+14): Machine 08f05013 unshfli x0,x0,15
# Info   MEMX 0x80005014 0x80005014 2 5013
# Info   MEMX 0x80005016 0x80005016 2 08f0
# Warning (RISCV_ILL) CPU 'refRoot/cpu' 0x80005014 08f05013 unshfli x0,x0,15: Illegal instruction - Zbp absent
# Info   mstatus 00004008 -> 00005880 [SD:0 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:2(Clean) MPP:0->3 VS:0(Off) SPP:0 MPIE:0->1 UBE:0 SPIE:0 MIE:1->0 SIE:0]
# Info   mepc 00000000 -> 80005014
# Info   mcause 00000000 -> 00000002 [Interrupt:0 Code:0(Instruction address misaligned)->2(Illegal instruction)]
# Info   mtval 00000000 -> 08f05013
# Info 24: 'refRoot/cpu', 0x0000000080000170(trap_handler_returnplus4): Machine 341022f3 csrrs   x5,mepc,x0
# Info   MEMX 0x80000170 0x80000170 2 22f3
# Info   MEMX 0x80000172 0x80000172 2 3410
# Info (OP_NCH) GlobalTime:0.000000 LocalTime:0.000000 Net:refRoot/readcsr 0 => 0x50341
# Info   x5 00004000 -> 80005014
# Info (IDV) Instruction executed prior to mismatch '0x80005010(main+10): 3002a073 csrrs   x0,mstatus,x5'
# Error (IDV) PC mismatch (HartId:0, PC:0x80000170 trap_handler_returnplus4+0):
# Error (IDV) Mismatch 0>
# Error (IDV)   . dut:0x80005014 main+14
# Error (IDV)   . ref:0x80000170 trap_handler_returnplus4+0
```

stineje, 2024-11-28:
Interesting; so, is this an issue with ImperasDV model.  If so, i believe they will be happy you found this.

davidharrishmc, 2024-11-30:
These issues appear to have been related to the rv32gc imperas.ic.  After tweaking the file for other reasons, ImperasDV matches.

closed 2024-11-30 by davidharrishmc