openhwfoundation/cvw #1161: Buildroot fails to compile on GCC 14

jordancarlin, 2024-12-01:
When the system (not riscv cross compiler) compiler is gcc14, buildroot fails to compile. I suspect this is because of new warnings/errors introduced by gcc 14 (we saw something similar with our crt0 code). 

With GCC 14 released and available as a package in many distros, this may start to be an issue. Especially with the next wave of releases when gcc14 is likely to become the default for some distros.

Need to investigate root cause/solution still, but I suspect moving to a newer version of buildroot would solve this.

davidharrishmc, 2024-12-02:
Yikes.  The textbook refers to certain statements and addresses in the buildroot binary.  When we update the version of build root, all that will change.  Probably it changes anyway with the new compiler.  I'm afraid we just need to indicate in the book that the examples came from a certain version and will differ from the latest version.

@jordancarlin do you want to experiment with whether newer buildroot solves the issue, and whether it still boots in lockstep?  This is a 2-day sim if all goes right and weeks or months of debug if not, so let's not commit anything until we know it actually works.

@rosethompson any thoughts?

jordancarlin, 2024-12-02:
Addresses will almost certainly change with different compilers even if we keep the same version of buildroot. We're already using different compilers on different distros so this seems kind of inevitable.

I can look into upgrading our buildroot version. Probably not a terrible idea anyway because I think newer buildroot and linux kernel supports several new RISC-V extensions that we have in the Wally buildroot configuration anyway.

jordancarlin, 2024-12-31:
Seems like there is a simpler, official way of storing our buildroot configuration that has much better compatibility across different versions of buildroot. I will look into storing our configuration in this format as part of upgrading to the newer version so that future upgrades are easier.

jordancarlin, 2024-12-31:
Upgrading our version of buildroot is also definitely going to be necessary in the nearish future because RHEL 10 (which just released in beta with a full release coming mid 2025) comes with GCC 14 as its default compiler.

jordancarlin, 2025-01-04:
Making good progress with this. The updated configuration method is working for buildroot with our existing version. This makes updating the version much easier.

I've gotten to printing the SBI banner using the new version of buildroot. Have been testing on VLSI because of space constraints on chips so no lockstep runs yet, but now that the new version of buildroot seems to be working I'll start a full lockstep boot soon before opening a PR.

jordancarlin, 2025-01-17:
Fixed with #1227

closed 2025-01-17 by jordancarlin