openhwfoundation/cvw #1171: Review wsim flags

jordancarlin, 2024-12-02:
Several of the options to wsim are either broken or of questionable value

- [ ] `--vcd`
  - Causes an error for Questa and seems to do nothing for Verilator. I believe it should be dumping a vcd file for viewing waves from Verilator and should probably not be allowed for the other simulators. Need to to investigate why it doesn’t actually produce the vcd dump from Verilator.
- [ ] `--locksteplog`
  - Purpose unclear. I know when I was adding `--lockstepverbose` there was discussion about what made it different from this but we never got to an answer. It’s also quite invasive in the wsim script and makes the lockstep logic significantly more convoluted.
- [ ] `--covlog`
  - Not currently used anywhere. Presumably just for debugging, but is this actually still useful?

rosethompson, 2024-12-02:
I would like --vcd to produce a vcd file for all simulators.  The vcd file can be used for more than just gtkwaves such as vector based power analysis using voltus.

locksteplog is an old version of lockstepverbose which can be removed.

I'm not sure what covlog is.

jordancarlin, 2024-12-02:
Sounds good. I'll remove locksteplog and look into why vcd is broken right now. Will try to get it working for all the simulators.

jordancarlin, 2024-12-02:
@davidharrishmc Any idea about covlog?

jordancarlin, 2024-12-02:
@rosethompson had to make a few changes to the testbench, but I got vcd dumps working for all the simulators. Something seems fishy with the Questa one though. It is an order of magnitude smaller than the dumps from Verilator and VCS...

Verilator: 4 GB
Questa: 9 MB
VCS: 6.3 GB

davidharrishmc, 2024-12-02:
I’ve never used covlog.  

I think Huda may have originally added it.  @Huda, do you still want this flag?

If you don’t have a use for it, we can take it out.

David

    parser.add_argument("--covlog", "-d", help="Log coverage after n instructions.", default=0)

rosethompson, 2024-12-02:
That's definitely wrong.  VCD files are very large.  What level of the device hierarchy is it logging?

Huda-10xe, 2024-12-03:
I never added the "covlog" flag, nor am I using it!!

jordancarlin, 2024-12-07:
Closing this because the flags have been fixed in #1177. Will open a new issue to track problems with VCD generation in Questa.

closed 2024-12-07 by jordancarlin