openhwfoundation/cvw #1176: Investigate a way to test python scripts for compatibility with all supported versions in install script

jordancarlin, 2024-12-07:
Depending on the distro being used, we install different versions of tools. For example, we install versions of python between 3.9 and 3.12. Investigate automated way of ensuring our scripts our compatible with all of these versions. This has already caused some issues for @rosethompson with python 3.12, and when working on the updated version of wsim I ran into the opposite issue and attempted to use a feature not present in python 3.9.

Potential solution is something like pylint.

jordancarlin, 2024-12-31:
Working on this in conjunction with the ongoing major cleanup/refactoring of many of the python scripts that I've been doing. Once that is complete, I will enable the Ruff linter in CI and have it lint for all versions of python from 3.9 onwards to ensure we don't have compatibility issues.

If we want to we can also enable pre-commit checks so that people can see the ruff lint failures before committing instead of just when the CI fails.

davidharrishmc, 2024-12-31:
Sounds great!

PR #1245 (2025-01-28): Python updates
- Refactor most of the python files to use more standard python conventions and conform to the ruff linter
- WIP: enable ruff lint in CI to ensure all python code is compatible with all needed python versions
  - Also make it possible to run this locally so errors can be discovered before pushing to GitHub

closes #1176
closed 2025-01-28 by davidharrishmc