openhwfoundation/cvw #1179: VCS fcov false support

jordancarlin, 2024-12-07:
`wsim` supports passing `--fcov` when using VCS and the `run_vcs` script accepts the `--fcov` flag, but it does not do anything with the flag so functional coverage is not actually generated. The `run_vcs` script either needs to be updated to properly support fcov or this flag should be disabled. @davidharrishmc not sure which path you want to take with this one.

davidharrishmc, 2024-12-08:
We aren’t using coverage in VCS yet.  It’s probably more than an hour to figure out how to add functional and code coverage, so I think it’s best to just print an unsupported error for now.

jordancarlin, 2024-12-08:
Code coverage is already set up. Just needed a quick fix that I pushed in #1177.

I'm making some changes to how fcov is handled by wsim in my next PR. I think with those changes it will be trivial to support it in VCS, but if that turns out not to be the case then I'll go ahead and have it print out an error instead of its current behavior of silently failing.

davidharrishmc, 2024-12-08:
If it works easily, that’s great and some users will be glad to have it.

Our code coverage depends on some TCL scripting to exclude uncoverable items.  It’s amazing to me that VCS runs the Questa script unchanged.  I’d think the vcover might need changing as well to merge all our coverage.

David

jordancarlin, 2024-12-08:
I haven’t looked at the report in all that much detail yet. I suspect the
exclusions aren’t working,  but it is at least collecting coverage again
instead of throwing a cryptic error message.

jordancarlin, 2025-04-18:
Disabling the option to pass `--ccov` or `--fcov` to `wsim` when using VCS and making this an enhancement.

closed 2026-11-06 by davidharrishmc