openhwfoundation/cvw #1181: Iterelf flag support

jordancarlin, 2024-12-09:
Currently `iterelf` accepts `--fcov` and `--coverage` flags and does nothing with either of them. We need to decide what the scope of iterelf should be (is it even all that useful anymore with the `addTestsByDir` function of `regression-wally`?) and update it accordingly.

davidharrishmc, 2024-12-09:
Iterelf coverage would be handy, and more so if it runs the jobs in
parallel and merges the results.it hasn’t risen to the priority because
regression-wally has been good enough, but if it’s a fast change, go for
it.

jordancarlin, 2024-12-09:
Ok. If you think it is worthwhile I can take a look. Just noticed that they were there but not doing anything so something definitely needs to change. If it seems like more trouble than it's worth I might just disable them.

davidharrishmc, 2024-12-09:
There have been several times when I've wanted iterelf for quick testing.  But without running in parallel, it's too slow to be regularly useful.

jordancarlin, 2024-12-10:
I can take a look, but I worry that updating it to run in parallel will force it to replicate a lot of the functionality already in regression-wally.