openhwfoundation/cvw #1186: Review multilibs being used for gcc

jordancarlin, 2024-12-09:
Right now we are building the following multilibs for gcc: 

- rv32e-ilp32e
- rv32i-ilp32
- rv32im-ilp32
- rv32iac-ilp32
- rv32imac-ilp32
- rv32imafc-ilp32f
- rv32imafdc-ilp32d
- rv64i-lp64
- rv64ic-lp64
- rv64iac-lp64
- rv64imac-lp64
- rv64imafdc-lp64d
- rv64im-lp64

Notably absent is anything related to bit manipulation, crypto, zifencei, and zfh, but many other extensions that we use are also missing. It is probably worth reviewing these and at least adding multilib configurations that match each of the main CVW configs.

davidharrishmc, 2024-12-09:
My understanding is that multilib relates to builds of the standard libraries.  I don’t think any of C standard libraries depend on crypto or Zifencei, and they probably work well enough without bit manipulation.  I’m inclined to leave this as it is, and a user with special requirements can customize their libraries.

It also isn’t clear that building with Zb* does anything right now.

https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1354

James knows more GCC than me and might have other ideas.

David

jordancarlin, 2024-12-10:
@stineje Any opinions/insight regarding multilibs? Otherwise I'm fine leaving as it is for now.