openhwfoundation/cvw #1214: PMP is too slow

davidharrishmc, 2024-12-31:
PMP is on the critical path, and only got worse after checking that the upper end of the access does not go beyond the top of the region.  It would be great to speed up PMP.  Perhaps some precomputations are possible.

davidharrishmc, 2025-04-13:
One solution would be to cache the PMP permissions for a page in the TLB entry.  If virtual memory is on and the TLB hits, PMP checking is fast.  Otherwise, add another cycle to do PMP checking.  If the page does not match a single PMP region, don't cache the PMP for that page, and check on the extra cycle.

PietDelaney, 2025-05-18:
Is this enhancement causing my build problem today?

(cvw) piet@piet-Precision-7780:/src/piet/cvw/cvw$ make all
make -C tests/riscof
mkdir -p riscof_work
mkdir -p work/riscv-arch-test
sed 's,{0},/src/piet/cvw/cvw.lattest/tests/riscof,g;s,{1},32e,g;s,{2},27,g' config.ini > config32e.ini
riscof run --work-dir=./riscof_work/arch32e --config=config32e.ini --suite=/src/piet/cvw/cvw/addins/riscv-arch-test/riscv-test-suite/ --env=/src/piet/cvw/cvw/addins/riscv-arch-test/riscv-test-suite/env --no-browser
    INFO | ****** RISCOF: RISC-V Architectural Test Framework 1.25.3 *******
    INFO | using riscv_isac version : 0.18.0
    INFO | using riscv_config version : 3.18.3
    INFO | Reading configuration from: /src/piet/cvw/cvw.lattest/tests/riscof/config32e.ini
    INFO | Preparing Models
    INFO | Input-ISA file
    INFO | ISACheck: Loading input file: /src/piet/cvw/cvw.lattest/tests/riscof/spike/spike_rv32e_isa.yaml
    INFO | ISACheck: Load Schema /opt/riscv/riscv-python/lib/python3.12/site-packages/riscv_config/schemas/schema_isa.yaml
    INFO | ISACheck: Processing Hart:0
    INFO | ISACheck: Initiating Validation for Hart:0
   ERROR | ISACheck: Error in /src/piet/cvw/cvw.lattest/tests/riscof/spike/spike_rv32e_isa.yaml.
   ERROR |  PMP:
   ERROR |   - unknown field
make[1]: *** [Makefile:30: arch32e] Error 1
rm config32e.ini
make: *** [Makefile:15: riscof] Error 2
(cvw) piet@piet-Precision-7780:/src/piet/cvw/cvw$

davidharrishmc, 2026-09-18:
This is a feature request for the future.  Closing for now.

closed 2026-09-18 by davidharrishmc