openhwfoundation/cvw #1263: New Buildroot FPGA bug

rosethompson, 2025-02-04:
The new Buildroot successfully boots on the vcu108 board and is able to run commands; however, it cannot switch to zsh.  zsh locks up the terminal and the software appears to be stuck in a trap storm (probably interrupt) as the SPI device has a pending interrupt which is never acknowledged. I don't have (m/scause) logged in the ILA.

jordancarlin, 2025-02-05:
Exciting that the new version is running on the FPGA! Strange issue with the interrupts. I wonder if it's configuration related with new version of SBI?

rosethompson, 2025-05-05:
I understand what is going on now. The above analysis has an error, please ignore.

This bug occurs when a misaligned memory access spans two pages and both cause DTLB misses and the second causes a D cache miss.  After the HPTW writes the first DTLB fill, the HPTW drives PAdrM with HPTWAdr; however, the LSU thinks the HPTW is done and allows the D cache to re-issue the original memory request.  In this manifestation a store double causes a memory fetch.  This fetch is then incorrectly writes into the cache's SRAM at PAdrM.  The HPTW begins walking the second page and gets a corrupted PTE.

The LSU prevents the D cache from re-issuing the original memory request by asserting SelHPTW.  The problem is after the first walk the HPTW FSM returns to IDLE lowering SelHPTW.  I think the solution is to find a simple way to continue to assert SelHPTW during this special case.

davidharrishmc, 2025-05-05:
This is great!  Is there a way to get cases like this into our functional verification?

rosethompson, 2025-05-06:
Partial good news. I created a test program that triggers this specific bug and a single line RTL solution that fixes the bug in simulation. However, the zsh still fails, but for a different reason. The first page table walk and first d cache access are correct.  The second page table walk gets to the last level of the table and reads a PTE of 0.  I'm not sure why this happens yet.

rosethompson, 2025-05-06:
I understand the second error now. The stval register contains the incorrect value.  We are recording the base virtual address of the store.  However, it is the spilled half which is missing from the page table. stval needs to be written with the address of the part of the access which causes the fault.  Quoted from  section 3.1 page 45 of the privilege spec 

> If mtval is written with a nonzero value when an instruction access-fault or page-fault exception
> occurs on a hart with variable-length instructions, then mtval will contain the virtual address of the
> portion of the instruction that caused the fault, while mepc will point to the beginning of the
> instruction.

There are probably similar bugs we haven't considered with xepc, and xcause.

davidharrishmc, 2025-05-06:
> rosethompson

Terrific!  Is there an easy solution?

rosethompson, 2025-05-06:
csr.sv
```
  always_comb
    if (InterruptM)           NextFaultMtvalM = '0;
    else case (CauseM)
      12, 1, 3:               NextFaultMtvalM = PCM;  // Instruction page/access faults, breakpoint
      2:                      NextFaultMtvalM = {{(P.XLEN-32){1'b0}}, InstrOrigM}; // Illegal instruction fault 
      0, 4, 6, 13, 15, 5, 7:  NextFaultMtvalM = IEUAdrM; // Instruction misaligned, Load/Store Misaligned/page/access faults
      default:                NextFaultMtvalM = '0; // Ecall, interrupts
    endcase
```

needs to replaced IEUAdrM with the IEUAdrSpillM generated in the LSU.

Similarly, PCM won't be the correct address for a instruction page fault spill. That one is bit more complex, because we currently don't pipeline the spilled PC.

rosethompson, 2025-05-07:
Fixed with PR #1404.

closed 2025-05-07 by rosethompson