openhwfoundation/cvw #1317: Testbench FSM stopping some Breker tests prematurely

jordancarlin, 2025-03-13:
Some of the Breker tests are being prematurely halted by this code in the wallyTracer:

```verilog
    if(HaltW) begin
`ifdef FCOV
      $display("Functional coverage test complete.");
`endif
`ifdef QUESTA
      $stop;  // if this is changed to $finish for Questa, wally.do does not go to the next step to run coverage and terminates without allowing GUI debug
`else
      $finish;
`endif
end
```

`HaltW` is driven by `HaltM`, which in turn is based on `DCacheFlushStart` from the testbench FSM.

davidharrishmc, 2025-03-13:
> jordancarlin
>  created an issue

DCacheFlushStart derives from TestComplete, which comes from

    TestComplete <= ((InstrM == 32'h6f) & dut.core.InstrValidM ) |
       ((dut.core.lsu.IEUAdrM == ProgramAddrLabelArray["tohost"] & dut.core.lsu.IEUAdrM != 0) & InstrMName == "SW"); // |

So it would appear that the Breker test is halting either due to a write to host or a jump to self causing an infinite loop.

jordancarlin, 2025-03-21:
Resolved by #1327. Turns out Trek was misconfigured and was generating tests for Sv57, which caused strange addressing behavior.

closed 2025-03-21 by jordancarlin