openhwfoundation/cvw #1324: Imperas lockstep simulation throws illegal instruction for pause instruction when Zifencei extension disabled

jordancarlin, 2025-03-20:
ImperasDV throws an illegal instruction exception when attempting a `pause` instruction when the Zifencei extension is disabled. The `pause` instruction is controlled by the Zihintpause extension, not Zifencei. Furthermore, even when Zihintpause is not disabled, the `pause` instruction should be decoded as a fence instruction with no successor set, and therefore be treated as a nop. It should never be treated as an illegal instruction.

davidharrishmc, 2025-03-20:
> (openhwgroup/cvw#1324) <https://github.com/openhwgroup/cvw/issues/1324>

Can you attach an elf and an imperas.ic to reproduce this?

snps-straus, 2025-03-21:
Actually, no test case needed. We see the issue and will add a new configuration option to enable just the zihintpause extension. Once that is done the pause instruction will only throw an exception if neither zihintpause nor zifencei are configured.

Thanks for pointing this out.

Jim Straus
Synopsys

davidharrishmc, 2025-03-21:
Excellent. Thank you!

davidharrishmc, 2025-04-24:
@eroom1966 is looking for an engineering build that fixes this.

snps-straus, 2025-04-25:
FYI, The fix went into the Imperas codebase today, so any build dated 20250325 or later will have the following fix (excerpted from CHANGELOG.MODEL.RISCV.txt):

> Fixes for the PAUSE instruction implementation
> -------------------------------------------------------------------------------
> - The Zifencei parameter no longer affects whether the PAUSE instruction is 
>   implemented. Previously the PAUSE instruction would throw an Illegal 
>   Instruction exception unless Zifencei was True. But since PAUSE is a special 
>   case of the required FENCE instruction it should always be allowed.
>   
> - A new parameter, Zihintpause, has been added to enable the PAUSE 
>   instruction, which is a special case of the FENCE instruction with rd=rs1=x0, 
>   fm=0, pred=W and succ=0.
>   When Zihintpause is True, disassembly of the special case FENCE instruction 
>   will be as a PAUSE instruction and, when the PAUSE_time_limit parameter is 
>   non-zero, executing a PAUSE will halt the processor in WFI mode for up to 
>   the specified number of cycles. When PAUSE_time_limit==0, PAUSE will be a 
>   NOP.
>   The default value for the Zihintpause parameter is True. 

Thanks for pointing out this issue, and for your patience.

Jim Straus
Synopsys

snps-straus, 2025-05-01:
@jordancarlin  An update with this fix has been sent. Please close this issue after confirming the fix, or let us know if you have any issues.

Thanks,

Jim

jordancarlin, 2025-05-01:
Great. Thanks for the update. I'll test it soon and let you know if there are further issues.

jordancarlin, 2025-05-08:
@snps-straus Just tested this with the new version, and everything seems to be working correctly. Thanks for the fix.

closed 2025-05-08 by jordancarlin