openhwfoundation/cvw #1350: Incorrect Exception Priority

C0d3ing, 2025-04-09:
Working on Ssstrict Csr test for supervisor mode and accidentally included the satp register. This caused a strange exception and led to a mismatch between IDV and Wally. According to the Priv spec exception code 4 should be returned when Zicclsm is not enabled. 

[image]

[SsstrictCsrS.elf.gz](https://github.com/user-attachments/files/19660253/SsstrictCsrS.elf.gz)

jordancarlin, 2025-04-09:
Interesting one. From section 3.1.15 of the spec:

> Load/store/AMO address-misaligned exceptions may have either higher or lower priority than load/store/AMO page-fault and access-fault exceptions.
>
> > The relative priority of load/store/AMO address-misaligned and page-fault exceptions is implementation-defined to flexibly cater to two design points. Implementations that never support misaligned accesses can unconditionally raise the misaligned-address exception without performing address translation or protection checks. Implementations that support misaligned accesses only to some physical addresses must translate and check the address before determining whether the misaligned access may proceed, in which case raising the page-fault exception or access is more appropriate.

So technically either exception is valid, but since rv32gc does not support misaligned accesses anywhere, it makes more sense to throw a misaligned exception.

davidharrishmc, 2025-04-09:
> jordancarlin

Yes, the language encourages our rv32gc configuration with no Zicclsm to unconditionally raise the misaligned-address exception.  That could happen without doing any address translation, if we want, but I think it’s harmless to leave the translation if that is easier.

davidharrishmc, 2025-11-09:
A root cause is that the MMU is not throwing LoadMisalignedFaultM because TLBMiss = 1. See mmu.sv at 145.  Either LoadMisalignedFault should be raised despite TLBMiss or LoadAccessFault should not be raised until TLBMiss is released.

[image]

Reproduce with 
wsim rv32gc ~/SsstrictCsrS.elf --lockstep --gui