openhwfoundation/cvw #1354: PMP interval bug

davidharrishmc, 2025-04-11:
PMP needs to be enhanced to match if any byte of an access is within the PMP interval, and to fail if any byte of the access is outside the interval.

This is a known bug caught by Breker or UET tests, but I don't see the issue posted, or a test case.

jordancarlin, 2025-04-11:
It is caught by the riscv-arch-test rv64 pmp tests (which I think were developed by UET). The current version of riscv-arch-test that we are using has these tests, but they are commented out in tests.vh.

The relevant tests are:
```
"rv64i_m/pmp/src/pmp64-TOR-M.S",
"rv64i_m/pmp/src/pmp64-TOR-S.S",
"rv64i_m/pmp/src/pmp64-TOR-U.S"
```

davidharrishmc, 2025-04-13:
I've partially fixed this - TOR top was not being checked against size. These riscv-arch-test pmp tests are now passing.

However, there still should be a PMP match when an 8-byte access includes a 4-byte NA4 or TOR region in the upper half of the access.  This isn't implemented correctly on Wally yet.

There also needs to be more work on PMP matches with misaligned accesses.  We must match if the region includes any of the bytes in the access, and fail unless they include all of the bytes.

lb = lower bound of access
Aub = upper bound of access
Rlb = lower bound of region
Rub = upper bound of region

Alb = physical address
Aub = physical address + size - 1

TOR
Rlb = pmpaddr_i-1 << 2
Rub = pmpaddr_i << 2 - 1

NAPOT
Rlb = pmpaddr masked << 2, all 0s
Rub = pmpaddr masked << 2, all 1s

MatchAllBytesOfAccess
Alb >= Rlb -> PA >= Rlb
Aub <= Rub -> PA + size - 1 <= Rub

MatchAnyByteOfAccess
Aub >= Rlb -> PA + size - 1 >= Rlb
Alb <= Rub -> PA <= Rub

Algorithm:
	Find lowest numbered PMP with MatchAnyByteOfAccess
	If ~MatchAllBytesOfAccess, then fail
	Else (L or mode == S or mode == U) then succeed if RWX matches access type
	else fail

davidharrishmc, 2025-06-03:
Resolved with #1433

closed 2025-06-03 by davidharrishmc