openhwfoundation/cvw #1380: Coverage tests fail to compile with latest binutils (2.44)

jordancarlin, 2025-04-23:
There is an alignment issue with some of the coverage tests when compiled with the latest binutils that prevents the final ELF from being produced. This needs to be fixed, and if we don't get it resolved before the 1.0 tag, we should make sure to pin the installation script to the older version of binutils.

rosethompson, 2025-04-24:
To clarify the non-coverage CLINT tests failed to build due to alignment.  The exception_vector_table of the WALLY-TEST-LIB-64.h (and 32.h) were in the .text segment rather than .data segment.  The latest linker exposed this bug.  The changes we made yesterday resolve the CLINT issues.

ZicsrS is the only coverage test failing for me.  It times out running code coverage after 1200 seconds.  It appears to be stuck in a trap loop. This occurs with binutils 2.44 and 2.43_2.

jordancarlin, 2025-05-13:
Tested with a fresh install and the wally-riscv-arch-test tests all compile fine now with binutils 2.44. Closing.

closed 2025-05-13 by jordancarlin