openhwfoundation/cvw #1388: Check GCC 15 compatibility

jordancarlin, 2025-04-26:
GCC 15 was just released. We need to either confirm that everything works correctly with it or pin the installation script to ensure it installs GCC 14 is there are issues with GCC 15 that aren't easy to resolve before the 1.0 release.

jordancarlin, 2025-05-20:
Installed gcc 15, and surprisingly the only things that fail to build are the zsbl and embench. All tests compile fine.

davidharrishmc, 2025-05-20:
The book examples are on 14.2.  Shall we pin it there?

jordancarlin, 2025-05-20:
We can definitely pin the versions of the tools installed by installation script for the 1.0 tag (see #1395). The fixes ended up being very easy though, so might as well make sure it is possible to compile things with the newer gcc release as well.

closed 2025-05-20 by jordancarlin