openhwfoundation/cvw #1397: Mismatch in InterruptsS test

HamzaYusufJamal, 2025-04-30:
wsim rv32gc InterruptsS_Mmode.elf --lockstep

```
# Info (IDV) Instruction executed prior to mismatch '0x80001120(resto_Mrtn+1c): 30200073 mret'
# Error (IDV) PC mismatch (HartId:0, PC:0x80000dc4 Mtrampoline+4):
# Error (IDV) Mismatch 0>
# Error (IDV)   . dut:0x80000664 cp_trigger_mti+24
# Error (IDV)   . ref:0x80000dc4 Mtrampoline+4
# Error (IDV) Insn. bit pattern mismatch (HartId:0, PC:0x80000dc4 Mtrampoline+4):
# Error (IDV) Mismatch 1>
# Error (IDV)   . dut:30047ff3 csrrci  x31,mstatus,8
# Error (IDV)   . ref:0880006f jal     x0,80000e4c
# Error (IDV) GPR register value mismatch (HartId:0, PC:0x80000dc4 Mtrampoline+4):
# Error (IDV) Mismatch 2> GPR x31
# Error (IDV)   . dut:0x00000088
# Error (IDV)   . ref:0x00001880 (not updated)
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x80000dc4 Mtrampoline+4):
# Error (IDV) Mismatch 3> CSR 300 (mstatus)
# Error (IDV)   . dut:0x00000080 SD:0 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0
# Error (IDV)   . ref:0x00001880 SD:0 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:3 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0
# Error (IDV) Mismatch 4> CSR 342 (mcause)
# Error (IDV)   . dut:0x80000007 Interrupt:1 Code:7(Machine timer interrupt) (not updated)
# Error (IDV)   . ref:0x80000001 Interrupt:1 Code:1(Supervisor software interrupt)
# Error (IDV) testbench.idv_trace2api.state_compare @ 897490: MISMATCH

davidharrishmc, 2025-04-30:
[InterruptsS.tar.gz](https://github.com/user-attachments/files/19985591/InterruptsS.tar.gz)

davidharrishmc, 2025-11-11:
This test case now throws an infinite loop but does not mismatch.  Something must have changed.  Closing issue.

closed 2025-11-11 by davidharrishmc