openhwfoundation/cvw #1472: Fail build script on ARM

doganulus, 2025-07-01:
The following lines in the build scripts fails on ARM builds (such as `ubuntu-latest-arm` runners) as they hardcode `x86_64` paths.

https://github.com/openhwgroup/cvw/blob/ef65739e21da18769822bfdf69513007941de299/bin/wally-package-install.sh#L159-L162

doganulus, 2025-07-01:
The Arm support is important in classrooms as many students run Arm-based machines (especially MacBooks). Since this repository targets educational programs, I would love if the support extends towards Arm.

doganulus, 2025-07-02:
We use Docker containers to standardize our computer architecture lab environment and distribute it to students over container registries. These containers work uniformly across Linux, Windows, and macOS, but we must be able to build them separately for each architecture.

I try to build the project in containers on Ubuntu ARM.

jordancarlin, 2025-07-03:
That will be much simpler to support. I will try to go through and switch all of the installation commands over to be architecture-agnostic in the next few days (or you're welcome to open a PR sooner). The trickier part will be Sail (which we download as a precompiled binary). I see that you've already opened https://github.com/rems-project/sail/pull/1369 though, so it might make sense to wait on that.

jordancarlin, 2025-10-15:
@doganulus Sail 0.20 has been released with a Linux ARM binary available. I'm not sure when I'll be able to get to this, but if you want to update the installation scripts to use it when running on ARM, I'd happily review a PR.

doganulus, 2025-10-17:
The problem is that installation scripts do more than what I consider sufficient and they are hard to modify. So I didn't use at the end. 

I could still try to help but I don't have the capacity during the semester.

closed 2025-10-26 by jordancarlin