openhwfoundation/cvw #1490: genesys2 synthesis not working

juanschroeder, 2025-07-22:
Hi there,

I'm trying to do the synthesis for genesys2 board like this:

`/fpga/generator$ make genesys2`

This gives the following errors:

```
INFO: [Synth 8-11241] undeclared symbol 'ahblite_resetn', assumed default net type 'wire' [/yocto/fpga/cvw/fpga/src/fpgaTopGenesys2.sv:212]
INFO: [Synth 8-11241] undeclared symbol 'cpu_reset', assumed default net type 'wire' [/yocto/fpga/cvw/fpga/src/fpgaTopGenesys2.sv:213]
INFO: [Synth 8-11241] undeclared symbol 'calib', assumed default net type 'wire' [/yocto/fpga/cvw/fpga/src/fpgaTopGenesys2.sv:214]
INFO: [Synth 8-11241] undeclared symbol 'phy_rx_dv', assumed default net type 'wire' [/yocto/fpga/cvw/fpga/src/fpgaTopGenesys2.sv:597]
INFO: [Synth 8-11241] undeclared symbol 'phy_rx_er', assumed default net type 'wire' [/yocto/fpga/cvw/fpga/src/fpgaTopGenesys2.sv:598]
INFO: [Synth 8-6157] synthesizing module 'fpgaTop' [/yocto/fpga/cvw/fpga/src/fpgaTopGenesys2.sv:31]
INFO: [Synth 8-6157] synthesizing module 'mmcm' [/yocto/fpga/cvw/fpga/generator/WallyFPGA.runs/synth_1/.Xil/Vivado-2090270-juan-Vivobook-ASUSLaptop-X1605VA-X1605VA/realtime/mmcm_stub.v:6]
INFO: [Synth 8-6155] done synthesizing module 'mmcm' (0#1) [/yocto/fpga/cvw/fpga/generator/WallyFPGA.runs/synth_1/.Xil/Vivado-2090270-juan-Vivobook-ASUSLaptop-X1605VA-X1605VA/realtime/mmcm_stub.v:6]
ERROR: [Synth 8-11365] for the instance 'mmcm' of module 'mmcm' declared at '/yocto/fpga/cvw/fpga/generator/WallyFPGA.runs/synth_1/.Xil/Vivado-2090270-juan-Vivobook-ASUSLaptop-X1605VA-X1605VA/realtime/mmcm_stub.v:6', named port connection 'clk_in1_p' does not exist [/yocto/fpga/cvw/fpga/src/fpgaTopGenesys2.sv:233]
ERROR: [Synth 8-11365] for the instance 'mmcm' of module 'mmcm' declared at '/yocto/fpga/cvw/fpga/generator/WallyFPGA.runs/synth_1/.Xil/Vivado-2090270-juan-Vivobook-ASUSLaptop-X1605VA-X1605VA/realtime/mmcm_stub.v:6', named port connection 'clk_in1_n' does not exist [/yocto/fpga/cvw/fpga/src/fpgaTopGenesys2.sv:234]
INFO: [Synth 8-6157] synthesizing module 'sysrst' [/yocto/fpga/cvw/fpga/generator/WallyFPGA.runs/synth_1/.Xil/Vivado-2090270-juan-Vivobook-ASUSLaptop-X1605VA-X1605VA/realtime/sysrst_stub.v:6]
ERROR: [Synth 8-6156] failed synthesizing module 'sysrst' [/yocto/fpga/cvw/fpga/generator/WallyFPGA.runs/synth_1/.Xil/Vivado-2090270-juan-Vivobook-ASUSLaptop-X1605VA-X1605VA/realtime/sysrst_stub.v:6]
ERROR: [Synth 8-6156] failed synthesizing module 'fpgaTop' [/yocto/fpga/cvw/fpga/src/fpgaTopGenesys2.sv:31]
---------------------------------------------------------------------------------
Finished RTL Elaboration : Time (s): cpu = 00:00:06 ; elapsed = 00:00:06 . Memory (MB): peak = 2196.000 ; gain = 539.766 ; free physical = 6204 ; free virtual = 15283
---------------------------------------------------------------------------------
RTL Elaboration failed
INFO: [Common 17-83] Releasing license: Synthesis
20 Infos, 30 Warnings, 1 Critical Warnings and 5 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
INFO: [Common 17-206] Exiting Vivado at Tue Jul 22 12:40:13 2025...
[Tue Jul 22 12:40:16 2025] synth_1 finished
ERROR: [Vivado 12-13638] Failed runs(s) : 'synth_1'
wait_on_runs: Time (s): cpu = 00:00:27 ; elapsed = 00:00:27 . Memory (MB): peak = 1493.656 ; gain = 0.000 ; free physical = 7609 ; free virtual = 16694
ERROR: [Common 17-39] 'wait_on_runs' failed due to earlier errors.

    while executing
"wait_on_run synth_1"
    (file "wally.tcl" line 86)
Vivado%

```

The Genesys 2 board has been integrated recently. Is this supposed to be working already?
I'm using Ubuntu 22.04 and Vivado 2024.2

Thanks

rosethompson, 2025-07-23:
I think I know what is wrong.  Did you first build for the Arty A7 and then build for the Genesys 2 board in the same directory?  The makefile has a bug. It is not smart enough to know the FPGA part has changed so it won't rebuild the AMD IP (mmcm, sysrst, etc).  

You must make clean if the board changes.  Can you try 
```
make clean
make genesys2
```

juanschroeder, 2025-07-24:
Thanks!

Indeed, it looks it was something that had been generated a few months ago when I had tried Wally previously and it was not cleaned up properly.

There is no 'clean' traget in 'fpga/generator'. I ended up removing 'IP' generated folder manually. But later I saw the make target to run was 'cleanAll'. The 'clean' target in the main folder did not (recursively) call this 'cleanAll' target in this folder either.

The bitstream was generated. I cannot really test it because I don't have a Genesys 2 board with me right now so I'll probably check if I can make it work in another Kintex-7 board I have.
Have you tried Linux in this board? That's what I'm actually interested in doing.

Thanks

juanschroeder, 2025-08-29:
I managed to make the Qmtech board fully boot Linux on Wally last week after observing that there were issues when accessing addresses > 128 MB (or so). I did so by using some kind of 'memtest' app (btw, it would be good if Wally's bootrom had such kind of test functionality, like it has for example LiteX 'BIOS' CLI).

After reducing Buildroot rootfs size (e.g. by removing all debug related stuff), the initramfs + Kernel went from 85 MB (or so) to 23 MB, which made the system boot without issues. **And I could check that this rootfs also made the Genesys 2 board boot Linux without issues.**

It doesn't seem to be a software issue but I didn't have time to check what could be wrong in the RTL.
I couldn't find any reference in the documentation to the ILA setup (debug .xdc files) which is also useful for 'bringup'.

davidharrishmc, 2025-08-29:
> juanschroeder

Interesting.  Glad you solved this.

The buildroot config produced from cvs/config/derivlist by running derivgen.pl specifies 2 GB of external memory.

EXT_MEM_SUPPORTED   1
EXT_MEM_BASE        64'h80000000
EXT_MEM_RANGE       64'h7FFFFFFF

The rv64gc config in cvw/config/rv64gc/config.vh specifies 128 MB of uncore memory (range = 7FFFFFF) and no external memory
(EXT_MEM_SUPPORTED = 0).

localparam logic UNCORE_RAM_SUPPORTED = 1;
localparam logic [63:0] UNCORE_RAM_BASE  = 64'h80000000;
localparam logic [63:0] UNCORE_RAM_RANGE = 64'h07FFFFFF;
localparam logic UNCORE_RAM_PRELOAD = 1'b0;
localparam logic EXT_MEM_SUPPORTED = 0;
localparam logic [63:0] EXT_MEM_BASE     = 64'h80000000;
localparam logic [63:0] EXT_MEM_RANGE    = 64'h07FFFFFF;

I haven’t spent time with the FPGA flow, but wonder if it is related to these config files.  Maybe buildroot still has UNCORE_RAM_SUPPORTED=1
and therefore hits both the uncore and external memory in the PMA for the first 128 MB, and then gets confused for the later part of external memory?

juanschroeder, 2026-01-03:
@rosethompson I created this PR: https://github.com/openhwgroup/cvw/pull/1601
If you think it could be useful and you want me to cleanup something let me know. I haven't really run any kind of tests beyond running the changes on the FPGA board itself.

I might try later to run Wally on the Nexys A7 board because I don't have the Arty A7.