openhwfoundation/cvw #1513: There is no exception when accessing unimplemented registers in RV32E mode

Pablotpa, 2025-08-25:
There is a problem I've gotten very frequently when verifying a configuration in RV32E mode. Since this config has only 16 registers, it should rise an illegal instruction exception when it tries to access any register in the 16-31 range, but instead it executes the instruction normally with what it seems a random value. For example, when executing mul x5, x4, x20, it doesn't rise an exception. Other case, c.addi x16, 0, it is decoded as a HINT although it should rise an exception since we are in E mode.

Pablotpa, 2025-08-26:
Thank you for your answer. I forgot to say that I'm already using a custom configuration (I added Zicsr among other extensions to rv32e.vh), so for other cases I get the expected exception but not for these cases. 
This is part of the config I'm using
```
// RV32 or RV64: XLEN = 32 or 64
localparam XLEN = 32'd32;

// IEEE 754 compliance
localparam logic IEEE754 = 0;

// RISC-V configuration per specification
// Base instruction set (defaults to I if E is not supported)
localparam logic E_SUPPORTED = 1;

// Integer instruction set extensions
localparam logic ZIFENCEI_SUPPORTED = 1; // Instruction-Fetch fence
localparam logic ZICSR_SUPPORTED    = 1; // CSR Instructions
localparam logic ZICCLSM_SUPPORTED  = 0; // Misaligned loads/stores
localparam logic ZICOND_SUPPORTED   = 0; // Integer conditional operations
```

davidharrishmc, 2025-08-26:
> Pablotpa

Then you probably found a real bug.  We’ve done very little testing of E. 
Would you like to modify controller.sv to detect illegal instructions when registers are out of bounds in E?
Make a PR when you are satisfied?

davidharrishmc, 2026-09-11:
The spec indicates "All encodings specifying the other registers x16-x31 are reserved."  Their behavior is UNSPECIFIED.  If Ssstrict is supported, reserved opcodes must trap.  However, this adds a good deal of hardware to the decoder, which counters the purpose of having a tiny E processor, so Wally is not implementing Ssstrict together with E unless there proves to be a compelling use case.

closed 2026-09-11 by davidharrishmc